BetaRavener/uPyLoader #1: keyword arg error

jknuteson, 2016-08-11:
Mint Linux 17.3  After pressing the Connect button ...
Traceback (most recent call last):
  File "main.py", line 199, in connect_pressed
    self.start_connection()
  File "main.py", line 182, in start_connection
    self._connection = SerialConnection(port, baud_rate, self._terminal)
  File "/home/jim/micropython/uPyLoader-master/src/serial_connection.py", line 23, in __init__
    self._serial = serial.Serial(self._port, self._baud_rate, timeout=0.2, write_timeout=0.2)
TypeError: __init__() got an unexpected keyword argument 'write_timeout'

profra, 2016-08-11:
@Beta_Ravener Hi, I made only a brief test... this tool seems the best among all I tried... very useful for fast work... definitely worth to continue work on it...
a few notes:
a/ please do refresh list of files on remote module after "Save to MCU" as you do after "Remove"
b/ I have more "stations" with uPy on internal net... they have AP disabled, each has own IP address... 
it would be nice to be able to connect with each... it means that your GUI with wifi connection should offer an IP address option... and remember last five-ten addresses :D
What is your opinion... and opinion of others?

BetaRavener, 2016-08-11:
@jknuteson Please, can you try using `pip list` to find out version of pyserial? 

It should be version >3.0. I have read that pyserial renamed the argument from `writeTimeout` in version 2, so it's possible that you have this old version. If true, you can use `pip install pyserial==3.1.1` to fix the issue and I will also update the instructions.

BetaRavener, 2016-08-11:
@profra Thanks for trying out! To keep this issue to its own, I have created new ones here: 
#2 for the refresh
#3 for choosing connection
If you find other things you'd like to see implemented, just open a new issue and select "Enhancement" label for it.

jknuteson, 2016-08-12:
Thanks @BetaRavener.  It was an issue with pyserial, which was pip'ed into python 2.7 rather than python 3.4.  I used `sudo apt-get install python3-pip` and checked that pyserial 3.1.1 was present in /usr/local/lib/python3.4/dist-packages/serial (again, I am using Mint 17.3).  

PyLoader is working and has connected to my ESP8266 (a Feather Hazzah) via the UART at 19200.  Excellent work @BetaRavener and nice Read.me directions!  Further testing to ensue after I flash the latest uPython firmware.  As an aside in my system with multiple versions of python, you must start PyLoader with` python3 main.py` 

I am closing this issue.

closed 2016-08-12 by jknuteson