BetaRavener/uPyLoader #5: Problems with installation under Linux Mint

profra, 2016-08-17:
See my output...

`fp@fp-VC62B ~/opt/uPyLoader $ python3 main.py
Traceback (most recent call last):
  File "main.py", line 15, in <module>
    from src.serial_connection import SerialConnection
  File "/home/fp/opt/uPyLoader/src/serial_connection.py", line 9, in <module>
    from src.setting import Settings
  File "/home/fp/opt/uPyLoader/src/setting.py", line 4, in <module>
    class Settings:
  File "/home/fp/opt/uPyLoader/src/setting.py", line 11, in Settings
    with open("config.txt") as file:
FileNotFoundError: [Errno 2] No such file or directory: 'config.txt'
`
... any advice?

BetaRavener, 2016-08-17:
It appears that configuration file was deleted by git pull. I have removed it in previous commit because every user will have different values (mainly wifi presets) in their config file, but I have not expected that the existing config files will be deleted. Anyway, new users (without previous `config.txt` would experience same error, so this needs to be fixed. Check in few minutes for fix.

profra, 2016-08-17:
Thanks, I am waiting... I have no problems with W10... you iterate to the fantastic tool :-) ... later I will come with some notes (enhacements)...

BetaRavener, 2016-08-17:
Please try to update by pulling again. Should be fixed by commit f7de090.

closed 2016-08-17 by BetaRavener