BetaRavener/uPyLoader #7: Flash from W10...

profra, 2016-08-23:
... I tried successfully but the output in flash window is not nice... see the embed image...... it would be nice to have the same output from uPyLoader as from esptool

[image]
)

BetaRavener, 2016-08-23:
Related to #6. Okay I see now why there were these special characters causing crash. Apparently, the esptool's output is not encoded in `utf-8` but in different encoding, probably depending on locale of the system it is running. I probably couldn't reproduce the error because I'm using English version of W10. @profra Could you run following code and report what it prints?
```
import sys
sys.stdout.encoding
```
Also, please run the esptool write_flash command like usual (as you did in the screen), but append `>output.txt` at the end. The command should be similar to this `esptool.py --port COM3 --baud 460800 write_flash --flash_size=8m 0 esp8266-20160809-v1.8.3-release.bin >output.txt`. Then, take the `output.txt` file from directory where you ran the command and without changing anything upload it [here](http://wikisend.com/). Then paste the download link in the comment. It will help me debug the issue. Thanks!

profra, 2016-08-23:
Task1
[image]

profra, 2016-08-23:
Task2... uploaded

BetaRavener, 2016-08-23:
@profra I have updated the first task to give me more clues and please run it under Python2 (the same as esptool is running under). 
And I can't see the download link anywhere for the file?

profra, 2016-08-23:
Task1_v2
[image]

profra, 2016-08-23:
Task2... I see download link in my comment... the comment has been edited twice...

BetaRavener, 2016-08-23:
@profra Can you confirm that the `output.txt` is not empty? The site is showing that it has size 13.1kb but when I try to download it, it's empty.

profra, 2016-08-23:
Sure not empty... once again
http://wikisend.com/download/665078/output.txt

BetaRavener, 2016-08-23:
Well, that file service seems to be broken, strange. Sorry for that, can you please send it as an attachment to dark-raven3@hotmail.com ? That should work..

profra, 2016-08-23:
Done

BetaRavener, 2016-08-23:
Well, now I'm confused. The file is exactly as expected with none of those special characters. Also, all the encodings are same as mine. If esptool has correct output, it can only get corrupted during piping (which is improbable), or during processing which would be strange to work in one case but not other. I will have to prepare some tests that you can run to identify the issue and then I will post back here.

BetaRavener, 2016-08-23:
@profra I have pushed new version with debug feature that should hopefully pinpoint the problem. You have to run the uPyLoader with a `--debug` switch, like so: `python3 main.py --debug`. Try flashing firmware with erase checked and if the text is corrupted again (which should be or else we are witnessing miracle), wait for it to finish and then upload `log.txt` from the uPyLoader folder here in the comments.

profra, 2016-08-23:
OK, I will start my NB with W10 now and I will let you know...

profra, 2016-08-23:
Done... 1 text file, 2 pictures
[log.txt](https://github.com/BetaRavener/uPyLoader/files/433421/log.txt)
[image]
[image]

BetaRavener, 2016-08-24:
@profra I have pushed new version, can you test it please? From the log file I learned that the data came through pipe correctly but somehow the processing routine must have corrupted them. The only place that I could think of (in those 10 or so lines) was statement that was deleting old characters. It was using `del` statement to remove them and maybe there memory was corrupted somehow. I replaced the `del` statement by array slicing which should work without any side effects. If the problem persists, again please upload the `log.txt` file so I can see if anything changed.

profra, 2016-08-24:
@BetaRavener OK, I will test it as soon as I can... I have to do something at work now...

profra, 2016-08-24:
@BetaRavener Done.... 1 file, 2 pictures (one with --debug, second without it) ...
... and it seems to be OK...THANKS!!!! ... please append on the end of every routine (Erase, Flash) the new line with word like "Done!" or so... one doesn't know for sure if it is END of process or has to wait for something other... we both know :-)

[log_2.txt](https://github.com/BetaRavener/uPyLoader/files/434812/log_2.txt)
[image]
[image]

BetaRavener, 2016-08-24:
*With relief* Glad that it's working, I will add those "done" messages  right away.

commit 0810d6ff (2016-08-24): Attempt to fix #7 and less flickering in flasher output.
closed 2016-08-24 by BetaRavener