BetaRavener/uPyLoader #9: Consider adding screenshot to README

pfalcon, 2016-10-31:
You posted a screenshot link in the topic on uPy forum, http://forum.micropython.org/viewtopic.php?f=16&t=2245 , and I would suggest to add something like that to README here on github. One in the forum (https://s9.postimg.org/ij6e9sqr3/u_Py_Loader.png) is way too big, so you may want to make a smaller window just for such screenshot. Here's an example how I do it: https://github.com/pfalcon/ScratchABit . Hope this simple improvement will allow to find more users for your tool.

BetaRavener, 2016-11-01:
Thanks for great suggestion, I'll definitely add it. It's true that people look for some visual clues first, so it will give more people an idea what's this repo about. Unfortunately, I have a lot of projects during semester so this one has to wait, but I do plan to improve it later. For now I can at least update the readme.

pfalcon, 2016-11-01:
Sure, there's no hurry. Thanks for your project.

BetaRavener, 2017-01-29:
Resolved.

closed 2017-01-29 by BetaRavener