BetaRavener/uPyLoader #13: After compile, move selection to the newly created .mpy file

jdeltoft, 2017-06-22:
This would make it one click then to just transfer instead of clicking to find the mpy file you just created.

BetaRavener, 2017-06-23:
There might as well be an option (either in settings or directly in main window) to automatically transfer `mpy` file after compilation. There may be some issues with this:
1) What's the name of compiled file? 
2) Did compilation finish successfully?

The first one I guess will always be the same as the file being compiled, just with `.mpy` extension, but I didn't verify that yet. The second should is related to #14, so if we don't detect any error it should be success. Otherwise process return code might be helpful. But this should be doable.

jdeltoft, 2017-06-23:
Yea, I think the mpy file name is always the same.  And yes, I agree.  Once there are more details on the compile finishing, then the file can be transferred only on success.

jdeltoft, 2017-06-27:
I'm starting to code this one up, and I'm stuck on being able to select the file in the local tree view.  I tried using:

self.localFilesTreeView.keyboardSearch(filename) 

but I can't seem to get that to work.  Any ideas?

BetaRavener, 2017-06-27:
I think what you want is the model, not view. That view is an instance of generic QTreeView that is able to display any tree-like structure, but the model is the one that holds data. You can access the model from view. It will be instance of QFileSystemModel. Then there's function that caught my attention: `index(const QString &path, int column = 0) const`. Through that you should be able to find index of the `.mpy` file. Then you just need to select item with that index. Easier said than done, I couldn't find any definitive answer but I would try to play with `setSelection` in view and creating `QItemSelectionModel` to use with it.

jdeltoft, 2017-06-28:
Ok, just did a pull request for my changes.  Thanks for the tips.  I eventually found what I needed (again, I'm a total noob to PyQt or even any Qt for that matter).  

https://stackoverflow.com/questions/4967660/selecting-a-row-in-qtreeview-programatically/10074855#10074855
https://stackoverflow.com/questions/17912812/how-do-i-select-a-row-in-a-qtreeview-programatically

jdeltoft, 2017-07-05:
This issue should be closed now, since it's covered by all the other changes in FileListImprovement and Autotransfer (that I eventually just merged into jdeltoft-master).  Closing for now.

closed 2017-07-05 by jdeltoft