BetaRavener/uPyLoader #17: Having a failure to transfer, but only SOME .mpy files??

jdeltoft, 2017-06-25:
So I've found a few cases where it seems that compiling the .py file to .mpy then causes failures in the transfer.  So far I have two such .py files.  Both start transferring and then when the progress bar is very much near the end, it says something like "failed to transfer".  Here is the code I put into "df.py" and have this issue.  Can you reproduce this?

`import gc
import os

def df():
  s = os.statvfs('//')
  return ('{0} MB'.format((s[0]*s[3])/1048576))

def free(full=False):
  F = gc.mem_free()
  A = gc.mem_alloc()
  T = F+A
  P = '{0:.2f}%'.format(F/T*100)
  if not full: return P
  else : return ('Total:{0} Free:{1} ({2})'.format(T,F,P))`

BetaRavener, 2017-06-26:
I will look into this today, but out of head there's no reason for any code to cause problems with transfer. WiFi is using websocket protocol to do that and UART uses transport scripts that just read the input and write it to file. While the script is running, the REPL on UART should be disabled so it shouldn't cause problems.

So just to check that I understood, it happens when trying to transfer compiled version of this file? There might be a some problem with encoding, although I don't see any special characters.

jdeltoft, 2017-06-26:
Yes, when I compile this file to a .mpy file, and transfer to the device it fails.  But the .py file transfers just fine, along with 6 or 7 other mpy files that all work fine.  I've now run into two mpy files that cause this, but this was the easiest one to share for you to see if you can replicated it.  Let me know what you think.  Thanks!

BetaRavener, 2017-06-26:
Ok ready to test it, but did you use UART or WiFi for transfer?

jdeltoft, 2017-06-26:
I used UART (I've never actually tried wifi)

jdeltoft, 2017-06-26:
Hmmm, now that I think about it, the two files that have issues both have portions of them cut and paste from a web browser.  Let me poke around to see what special (invisible) characters I may have in the files.

BetaRavener, 2017-06-26:
I just reproduced the problem.. at first it transferred ok because I fixed indent and moved those returns on separate lines from condition. Then I placed them back on the same line and got error during transfer. The code itself doesn't have any special characters but there might be something in compiled file that causes this. Going to look into it.

jdeltoft, 2017-06-26:
ok, thanks.  I tried dos2unix and unix2dos.  Both failed, which I now expect based on your comment.

BetaRavener, 2017-06-26:
Well, debugging the serial port it seems that there is some unescaped sequence because I get traceback from KeyboardInterupt exception. The problem is that even though the script is running, REPL is listening for those CTRL-C and similar sequences on the same UART as transfer is going, so transfering these special bytes causes unwanted behavior. There is already escaping mechanism that should prevent this, however it seems that some byte value is still causing this to happen.

BetaRavener, 2017-06-26:
Beat me to it. It's not the content that is causing exception but the protocol. It's really simple - each chunk starts with `#` character followed by one byte specifying chunk size (0-255). And as you can probably already see, when that size is for example 3, it gets interpreted by REPL as CTRL-C. The simple fix would be making highest bit always 1, changing the interval to (128-255) because the chunk produced by uPyLoader is anyway 64 bytes. I'm just trying to come up with solution that wouldn't require changes in transfer scripts, because everyone will need to reupload them.

jdeltoft, 2017-06-26:
Personally I wouldn't mind reuploading the transfer scripts.  They're tiny and it's easy to do from the tool. But that being said, I can see how it would be hard maybe to communicate to everyone they needed to update these.  Who reads the README.  :)

BetaRavener, 2017-06-26:
Ok I have fixed the issue by modifying script and actually created skeleton for version checking of those 2 files. If I ever need to make more changes, the user will be at least notified of mismatch between uPyLoader and scripts versions. The actual checking already works (done at every device connect through UART) but I need to propagate the result and display a message box. This should also notify if the scripts are actually missing and user has those scripts enabled in settings.

The downside is little longer connection time to accommodate all the communication and timeouts.

BetaRavener, 2017-06-26:
Please, check if the problem was solved and new ones not introduced :) For me the transfer of that `.mpy` file is working.

jdeltoft, 2017-06-27:
Ok, I had to manually delete the uploader/downloader.py files on my device though, and reInit them.  I assume you expected that?

BetaRavener, 2017-06-27:
Not really, the old scripts should have been overwritten. I've just tested by placing random garbage in one of them and sending new scripts. It overwrote the file without problem.

BetaRavener, 2017-06-27:
The version checking is now working too.
[image]

jdeltoft, 2017-06-27:
Ok, I have a two more devices I can test later that still have the old versions in them.

jdeltoft, 2017-06-27:
Ok, tested on my old dev board that had the old versions.  It worked great.  I wasn't expecting it at the first time start up, but rather when I tried to start a transfer.  But this is better!  Anyway, I did the Init transfer files, and then transferred all my mpy files (even the two bad ones now worked of course too).  Thanks!  I'll close this one out.

closed 2017-06-27 by jdeltoft