BetaRavener/uPyLoader #21: Uncommited settings.ui file

BetaRavener, 2017-07-05:
@jdeltoft Hi, could you please commit `settings.ui` file from `gui/qt` folder? You probably just committed the built `.py` file, because when I rebuilt the gui, I get missing attribute error on SettingsDialog. Thanks :)

jdeltoft, 2017-07-05:
@BetaRavener well, here's where I have to mention again that I had zero experience with PyQt when I was making my updates.  Embarrassingly I made my changes to the py files by hand, since I wasn't aware of the buildgui.py or how it worked.  Is there something I can quickly update to clean this up?

BetaRavener, 2017-07-05:
It's not a problem, jumping on the ship without any documentation can be really hard :) But we can leave that to another conversation. To resolve this: 

Normally what I do is edit `.ui` files in QtCreator (more specifically QtDesigner but I like to get the whole package, it's not that big). You open the `.ui` file there and you can change the GUI in graphical way. When finished, you just save the `.ui` file, call `buildgui.py` script and you're done. Then you just inherit from the class that was generated and build logic around it. I'm going to make those changes now so I don't have to resolve merge conflicts ;)

BetaRavener, 2017-07-05:
Fixed in commit  7c9f738.

closed 2017-07-05 by BetaRavener

jdeltoft, 2017-07-05:
Great to know about the QtCreator, thanks for the tip.  I definitely did it the *hard* way then.  LOL.  But hey, I suppose I did learn a bit doing it that way (or that's what I'll tell myself).   Thanks for cleaning up my mess.