BetaRavener/uPyLoader #24: File->Init Transfer Files  Issue

BOB63, 2017-07-09:
Hi, 
I'm trying to use uPyLoader 0.1.1 on W10 machine and a ESP8266 on which I've already flashed Micropython .
I'm able to connect the board via serial port and on the right window I've listed the two files  in the board but not transfer on my pc .
At startup I got message that UART files are missing and doing File->Init Transfer Files  Issue  the program is in stall. 
On the terminal window when I try to transfer from board to PC a file i got this error msg:
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== file_name="boot.py"
=== with open("__download.py") as f:
===     exec(f.read(), globals())
=== 
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
OSError: [Errno 2] ENOENT
>>> #3#0#

Do you have any suggestion ?
Thanks.

BetaRavener, 2017-07-09:
Hi, no problem with the font. Let me just double-check what you did:
1. Connected to ESP via UART (serial)
2. Got warning message about transfer files
3. Tried to `File->Init Transfer Files` **but the program stalled/hung**?
4. Proceeded anyway to transfer a file to ESP (should be named `boot.py`.

If it's so, then the problem is obviously in third step. File transfer is dependent on the files that are copied in step 3. Without them, you get the error (ENOENT) because the transfer files do not exist on the ESP .

To fix this, you have to make sure that `__download.py` and `__upload.py` are on the ESP and you no longer get that warning message when connecting to board. If uPyLoader hangs during initialization, please try again. The problem is that without these scripts, file transfer is unreliable (that's why they are used in first place) so errors may occur.

BetaRavener, 2017-07-09:
The initialization can be made more reliable by sending only `__upload.py` and using reliable transfer for sending `__download.py`. Error checking and timeouts can be used to detect when transfer failed (to output message) and prevent hang ups.

BOB63, 2017-07-09:
Hi , are the upload.py and download.py the two files that I find into the MCU folder of the code or where can I find those files ?

BetaRavener, 2017-07-09:
Yes, these are the scripts that will be transferred BUT you can't transfer them yourself. You have to use `File -> Init Transfer Files`. If everything goes right, you should see those 2 files listed on the connected device as in the image below.

[image]

BOB63, 2017-07-09:
No, unfortunately this option doesn't work . File transfer windows don't show any progress and I've to force the closing of program.

jdeltoft, 2017-07-09:
Are you using com or wifi?   Are you sure you have the right com port?

BOB63, 2017-07-10:
I'm using com. As soon as I push the connect button on the right window I see the files on esp8266, so I suppose that the connection is OK .

Inviato da smartphone Samsung Galaxy.
-------- Messaggio originale --------Da: Justin Eltoft <notifications@github.com> Data: 10/07/17  00:01  (GMT+01:00) A: BetaRavener/uPyLoader <uPyLoader@noreply.github.com> Cc: Roberto Portesani <speedlight86@gmail.com>, Author <author@noreply.github.com> Oggetto: Re: [BetaRavener/uPyLoader] File->Init Transfer Files Issue (#24) 
Are you using com or wifi?   Are you sure you have the right com port?

BetaRavener, 2017-07-10:
Ok sorry about that. I can now reproduce the problem and will be fixing it today. The problem was that the transfer scripts were not packed with pyinstaller that creates executable. The reason why we didn't see that happening is because we have source code downloaded and executable in the same folder, so our executable would pull these transfer files from source code folder.

BetaRavener, 2017-07-10:
The issue should be fixed in source code and I will release executables in few moments. Please let me know if the problem is gone in release 0.1.2 when you see it available.

BOB63, 2017-07-10:
Great job !!  Now via serial interface it works. I've to test via wifi because it doesn't work, but I'm facing some problems also with Webrepl , so may be something wrong in my setting .
Anyway thank again for now . I'll keep you informed if I find other issues .
Thanks. Roberto.

BOB63, 2017-07-10:
Also wifii works, was just the setting of my antivirus that was blocking the connection.

BetaRavener, 2017-07-10:
Ok great, closing the issue.

closed 2017-07-10 by BetaRavener

xpiolx, 2018-03-07:
This function stop working with the new release 0.1.3. Tried in 2 PC (both Win10 64b, esp1.9.3) and only v0.1.2 work.

BetaRavener, 2018-03-07:
@xpiolx Thanks for reporting an issue, but I will need more information. What exactly is your problem? Uploading transfer scripts to MCU? What is the behavior (something getting stuck, crashing, no visible change,...)? Is the connection to MCU working without problems (files listed correctly on connect, terminal responding to commands)? 

I'm not 100% sure, but few days ago I had to download release 0.1.3 and run it in new folder, and I think I did upload the scripts without problem, so it might be an individual problem.

BetaRavener, 2018-03-08:
`v0.1.2: Normal, recognizes that there is no files for upload/download on the board` this is incorrect. In v0.1.2, there was no checking for transfer scripts. It was added in v0.1.3 and it is precisely the message you see after connecting.

The transfer scripts and also uPyLoader code which handles transfers changed significantly from 0.1.2 to 0.1.3. This was because 0.1.2 used custom message format that was problematic on some firmwares.

There might be an issue when coming from v0.1.2 to v0.1.3, if the old transfer scripts are on the MCU. You should try delete those first if you didn't already. Other than that, this would require debugging on your side. `File transfer failed` popping up **imediately** after `File->Init` is not expected behavior. You copied the terminal content after trying to read a file `boot.py`, but I would be more interested if there is any output during `File->Init`. There certainly should be something.

Also, if you are more capable, you could always try pulling the latest source code and run uPyLoader in debugger. The function responsible for uploading transfer scripts is [this one](https://github.com/BetaRavener/uPyLoader/blob/master/src/connection/serial_connection.py#L213), so you should set breakpoint at its first line and step each line. Don't take too long, there are some timeouts that would also cause file transfer to fail, you're just trying to identify where it goes bad. Another function which this one calls is `send_file`, so if it fails on that line, you should step into this other function to get more details.

I know it's not your responsibility as a user but since this is just a side project for me, I could use some help :) Hope we can solve this one somehow together. I could also try to add some diagnostics in next release, but again, it takes time and I'm currently short on it.

mindforger, 2019-04-23:
i got a similar issue on my w10 system but i am not sure if this is an issue with my device since it is a snowflake device

i just managed to flash a specialised version of micropython to an NUCLEO-L432KC board with a minimalistic FS of only 23kBytes (i further improved it to 28k by removing some socket related stuff that was left behind, as it has no wifi or anything besides an frozen sdcard.py module to attach an sd)

when i want to init transfer files i get to around 9% and then get the described file transfer failed message

why exactly am i not allowed to manually transfer it using a temporary linux environment?

i got tools to copy over the files myself but using them is a pain and even if i move and name those files like described, the tool won't accept those files, it feels just wrong that you are forced to use the methode that is not working for whatever reason

i don't know how to run python in debug on my mini VM, its very limited and can not access internet because of firewall reasons and sharing files between host and VM is also a pain to do too, so i am desperate to use uPyLoader for windows but i can not get the initial file transfer files working 

if i could freeze them it would be even better for me but that failed catastrophically as my micropython just hangs up completely when i want to transfer a file

BetaRavener, 2019-04-23:
I'm not sure I really get you. First of all, you should be able to upload transfer scripts yourself - they are in the `mcu` folder. There isn't any check if you used uPyLoader to get them there and I don't modify them in any way during transfer. The only check is for the version, which is first line comment and will be there if you transfer them yourself anyway.

If you want to use uPyLoader without transfer scripts, there's also hidden setting that you can change in `config.json` called `use_transfer_scripts`. Normally, it's set to `true`, so just change it to `false`. It's a bit of legacy code but should still work. uPyLoader then uses paste mode which might be unreliable, but should be fine for small files.

The problem with freezing is that the files currently have to be edited (their code) before they are run. I wonder why I designed it in such way as there might be a better solution, so maybe I'll have look on overhauling that aspect.

mindforger, 2019-04-23:
> If you want to use uPyLoader without transfer scripts, there's also hidden setting that you can change in config.json called use_transfer_scripts. Normally, it's set to true, so just change it to [...]

where is that config.json located? i am using the compiled windows binary because running python behind this damn ssl-box is a major pain