BetaRavener/uPyLoader #27: closeEvent in main.py not called from binary exe version?

jdeltoft, 2017-09-08:
I noticed that when I check the "auto transfer" checkbox on the main screen, it isn't preserved for me through closing and reopening uPyLoader.  I thought it was just not implemented and was going to implement it myself when I noticed something interesting.  It is implemented, and works when I use "python main.py" to start the application, and click the window manager X (windows OS) to close the app.  But I then tried the windows exe and sure enough it's not working.  

So that probably means all that other code in closeEvent() isn't being handled either perhaps?  Or is the settings file just not given time to close in the case of the exe version for some reason?

jdeltoft, 2017-09-08:
Ok, things get stranger.  I rebuilt the uPyLoader.exe myself with the normal command:

**upyinstaller main.spec**

and then the exe worked fine to preserve the state of auto_transfer.  But when I looked at the blame on main.py, it looked like the changes were from 2 months ago.  So I need to see when the windows binary release file was created but I don't see how/why it wouldn't include the latest.  Is there something about the environment you're using to build the exe for windows?

jdeltoft, 2017-09-08:
Ok, I think I see what my problem is now.  If in windows you mark the config.json file as "hidden" then it no longer is able to update/write settings to that file, but it *does* read from them just fine!

BetaRavener, 2017-09-08:
Are you sure that the file doesn't become read-only? Also, debugging the closeEvent() might reveal some exceptions happening, that possibly crash the program before settings are saved, but you don't acknowledge them because you except the app to close anyway.

BetaRavener, 2017-10-06:
Resolved by #29.

closed 2017-10-06 by BetaRavener