BetaRavener/uPyLoader #31: Size of name column in tree view

BetaRavener, 2017-11-08:
Decide what behavior should name column have. It's nice when it resizes to content (ResizeToContent flag), but this makes it fixed width so that users can't change it. Options:
* Resize column to content after navigating to other directory, but remain resizable.
* Resize to content but stay the same width once user modifies the width (even after program closes).
* Always keep width set by user, but have a button (/something) in GUI through which user can resize to content on demand.

Can be also solved by adding option to settings for preferred behavior.

profra, 2017-11-08:
+1 for "Always keep width set by the user, but have a button (/something) in GUI through which user can resize content on demand. "

jdeltoft, 2017-11-20:
I'm confused by option 2.  At what point would you recent to content in that case?  Since you say it remembers my modified size even after program closes, then if I've ever changed the size, wouldn't this always prevent a resize to content?

I would vote for option 3 **if you said it was like excel** and a double click did the auto sizing.  I don't like adding another button in the UI for resize to content.  Too much clutter.

BetaRavener, 2017-11-20:
@jdeltoft Ad 2) - yes, once you ever set the width, it won't go back to autoresize. I can see problem there with user accidentally changing width, but it was just an option.

I like the double-click header idea, although I can imagine that many people wouldn't know about it, so an additional `View->Resize to Contents` item in menu might be a good idea. I'm also not sure how easy catching the double-click event will be, as normally you don't get access to header items.

jdeltoft, 2017-11-20:
Yea, I wasn't sure if it would even be possible in pyQT but figured I would mention it, and I think you'd be surprised how many people may know about the double click already.  Excel and sheets from google both support it.  But I also do agree maybe a preference check would be good.  Actually, that may solve the question about item 2.  If you select that check box, it auto sizes until you ever grab it again and change it.  Then the checkbox is unchecked.

BetaRavener, 2017-11-26:
Found some time to work on this. It was mostly searching web for how to do this, but it seems to be working as proposed :) Only problem that I couldn't overcome is that

BetaRavener, 2017-12-01:
The "intentionally reverse" thing won't work if the double-clicked header item was not the one by which the contents were originally sorted, so it's better to just leave it out. The proper implementation would require subclassing` QHeaderView` or using `eventFilter` to properly distinguish between events, which is not worth the time if people don't get mad because of it. If someone does, just reopen this issue and I'll look into it.

closed 2017-12-01 by BetaRavener