BetaRavener/uPyLoader #33: preferred port doesn't work for me on Mac

jdeltoft, 2017-11-15:
My port on the new mac I'm testing on is /dev/tty.SLAB_USBtoUART and it's not working if I put that as my preference setting for preferred port.  Hitting refresh always selects /dev/tty.Bluetooth-Incomming-Port even though the slab usb one is there second in the list.

BetaRavener, 2017-11-15:
Well, there's currently some strange logic that causes the search to fail. I will try to fix it while respecting what the current code was probably trying to do.

BetaRavener, 2017-11-15:
Fixed by 8b18ad2be90ab3aa020813ba0070a3559ab65ff2 in master branch. Please check that it's working correctly now.

jdeltoft, 2017-11-16:
I'll test it out tonight when I'm back at my computer.

jdeltoft, 2017-11-18:
Yes, this works!  thanks for fixing it.  Per my comments, I have seriously no idea what I was doing before with that split/join???  LOL!

closed 2017-11-18 by jdeltoft