BetaRavener/uPyLoader #34: Mac executable not working

jdeltoft, 2017-11-19:
I built on mac with pyinstaller, and it gave no errors.  But then when I updated the permissions to execute the output in dist folder it gave the following error.  I haven't had time to dig in, but wanted to at least post here for starting a thread on it.

justins-mbp:dist justin$ ls
uPyLoader
justins-mbp:dist justin$ chmod 755 uPyLoader 
justins-mbp:dist justin$ ./uPyLoader 
Traceback (most recent call last):
  File "main.py", line 29, in <module>
  File "src/gui/main_window.py", line 89, in __init__
  File "src/gui/main_window.py", line 167, in disconnected
  File "src/gui/main_window.py", line 115, in refresh_ports
AttributeError: 'NoneType' object has no attribute 'upper'
[586] Failed to execute script main
justins-mbp:dist justin$

BetaRavener, 2017-11-19:
Well the error is due to executing `upper` on `None`. So we know it should be string, but isn't. This happens in `refresh_ports`, which is responsible for populating the combo-box with connection options. My guess is that the Preferred Port in Settings is `None`, which seems plausible if you are running for first time. As such, it can't be compared to anything and trying to do so results in this exception. Fix is simple and will be ready in few minutes.

jdeltoft, 2017-11-20:
will test it out tonight.  thanks!

jdeltoft, 2017-11-21:
Still fails after a git fetch/pull/rebuild.  Same error, same spot.

BetaRavener, 2017-11-21:
Please, try to debug `refresh_ports`, more specifically its call to `scan_connections` in `self._connection_scanner.scan_connections(with_wifi=True)`. You should easily recognize where the problem is (list is being filled with `None` etc.).

jdeltoft, 2017-11-22:
Here is the pdb output for this area.  So I'm suspecting that the path to the preferences isn't working somehow.  It's not finding them (since preferred port is None) unless you start up uPyLoader in the same path as those json preferences

> /Users/justin/uPyLoader/src/gui/main_window.py(115)refresh_ports()
-> if self._connection_scanner.port_list:
(Pdb) l
110  	        self._connection_scanner.scan_connections(with_wifi=True)
111  	        # Populate port combo box and select default
112  	        self.connectionComboBox.clear()
113  	
114  	        pdb.set_trace()
115  ->	        if self._connection_scanner.port_list:
116  	            selected_port_idx = 0
117  	            pref_port = Settings().preferred_port.upper()
118  	            for i, port in enumerate(self._connection_scanner.port_list):
119  	                self.connectionComboBox.addItem(port)
120  	                if pref_port and port.upper() == pref_port:
(Pdb) self._connection_scanner.port_list
['/dev/tty.Bluetooth-Incoming-Port', '/dev/tty.SLAB_USBtoUART', 'wifi']
(Pdb) n
> /Users/justin/uPyLoader/src/gui/main_window.py(116)refresh_ports()
-> selected_port_idx = 0
(Pdb) n
> /Users/justin/uPyLoader/src/gui/main_window.py(117)refresh_ports()
-> pref_port = Settings().preferred_port.upper()
(Pdb) n
AttributeError: 'NoneType' object has no attribute 'upper'
> /Users/justin/uPyLoader/src/gui/main_window.py(117)refresh_ports()
-> pref_port = Settings().preferred_port.upper()
(Pdb) Settings().preferred_port
(Pdb) Settings()
<src.utility.settings.Settings object at 0x10b240278>
(Pdb)

BetaRavener, 2017-11-22:
Ah crap, I can see it now.. By mistake, I have put the condition checking if pref_port is `None` in wrong place. It should be here `pref_port = Settings().preferred_port.upper()`. I just probably looked where the `upper` is and didn't think much about it. So it's caused exactly by what I was thinking, I just didn't fix it (yet).

And regarding "unless you start up uPyLoader in the same path as those json preferences", yes of course - I have been telling this few times already. uPL is using it's current working directory to search for resources. So if you run it from different directory, it can't find them. It could be solved by creating uPL folder in user folder (`AppData\Roaming` on Windows) and storing settings there, but it's just as easy to create shortcut to uPL and set the working directory to its folder. You can then place the shortcut anywhere you want and also run it from anywhere.

jdeltoft, 2017-11-22:
Yea, I do know about the config and path dependency.  :-)  but in this case why isn't a new config.json created?  I'll have to play around and look at the code in that case.    Of course as you said, either way the pref port needs to be checked for None

jdeltoft, 2017-11-22:
I'm not as familiar with mac osx so I'll have to see why the exec icon doesn't allow me to run it either when I click to start uPyLoader from the icon.  The screen goes away too fast, so I'll see what else I can do.  Starting here:  https://pythonhosted.org/PyInstaller/when-things-go-wrong.html

jdeltoft, 2017-11-22:
Some other references.

 https://stackoverflow.com/questions/22418418/osx-app-built-with-python-quits-immediately-if-app-bundle-is-executed-from-finde

https://stackoverflow.com/questions/26176169/pyinstaller-os-x-app-runs-from-command-line-but-not-finder-window

jdeltoft, 2017-11-22:
Also per this comment sounds like maybe I can get to see the error message in the terminal still with "open":

- You can use the open command in a shell script to simulate double-clicking on a file in the Finder.

BetaRavener, 2017-11-22:
"But in this case why isn't a new config.json created?"

Because Settings are saved (and thus file created) when application closes nicely (e.g. user closing it). If it crashes, how it's supposed to create that file? :) To solve your suffering, I'm going to publish (the correct) fix.

BetaRavener, 2017-11-22:
Anyway, it there really is problem with relative paths under OSX, I will try to fix it later. But uPL shouldn't crash on start because of that - if settings file is not found, default one are used, and if icons are not found, there will be just blank space (although they are bundled directly in the executable so this shouldn't be an issue).

jdeltoft, 2017-11-22:
ok, sounds good!

jdeltoft, 2017-11-23:
Ok, this works now.  I have a new problem though, I'll enter a different CR

closed 2017-11-23 by jdeltoft