BetaRavener/uPyLoader #37: Progress bar and file size

jdeltoft, 2017-11-22:
I had a 450Kbyte log.txt file on the node that I was pulling off.  It took a LONG time to pull, and I had no idea if it was going well or not.  Would be nice to list out file size, and have progress bar on this transfer of log files from node to pc.

BetaRavener, 2017-11-23:
I have already tried to do that, but there's one problem - if you want to know file size, you first have to read it. Then you have to read it again when actually transferring the contents. I didn't want to make transfer scripts more complicated or taking more resources, so I bagged the idea. 

Anyway, I think using WiFi connection to ESP will be much faster, because there are routines in firmware for sending and receiving files.

jdeltoft, 2017-11-23:
Could you do something like a CRC calculation when you're writing files to the node, and every time you ever write a file (or remove a file) this CRC file gets updated?  Then you would only need to read a small "tags" type of file out.

BetaRavener, 2017-11-23:
Of course I can do everything.. the question is - should I? It's lot of work, just for having progress bar that is useful for about 1% of usecases. I don't think there's lot of people moving huge files around all the time (not through slow UART anyway). If I had to implement it, I would just read the file and count bytes like I stated above. Spamming files around MCU isn't good idea. 

Anyway, I currently don't have time to work on uPL as much, so things like this won't get my attention. I'm always looking forward to a working pull request if someone wants to help with development.

jdeltoft, 2017-11-24:
No worries.  I just figured I would put the issue out there.  I don't have time to fix either right now but maybe when I start transferring more large files.

BetaRavener, 2018-03-30:
I just realized there's an easier way to do this. There's `os.stat` command that uses filesystem information to get size of the file without reading it. If I ran this before transferring file, I'd be able to report progress correctly. I'll place it on my todo list and get back to it when I'll have some time.

BetaRavener, 2018-07-18:
Should be working both for UART and WiFi.

closed 2018-07-18 by BetaRavener