BetaRavener/uPyLoader #38: On Mac, the default folder for "app" is "/" which isn't writeable for config.json

jdeltoft, 2017-11-23:
I added the following to the main.spec file:

app = BUNDLE(exe,
          name='uPyLoader.app',
          icon='icons//main.ico',
          bundle_identifier=None)

based on this info page:  https://pythonhosted.org/PyInstaller/spec-files.html

While this makes a mac OSX app file directory, which works to not open a console, when I run that app, and update the navigation path for example, and then quit normally there is no config.json to be found.  And sure enough, next time I launch it doesn't have any config data to load.  When I start up uPyLoader I notice that the path listed for PC is "/" so that indicates I think that it would try to write config.json there which it won't have permission for.  Can we make it ~/ ?

BetaRavener, 2017-11-23:
How would you make it `~/`? I honestly don't know how. I think the only solution is to convert relative paths into absolute, as written in one of those SO posts.

BetaRavener, 2017-11-23:
I created `dev-mac` branch for solving these issues and pushed one commit in which I convert relative path of `config.json` to absolute. You should pull that branch and try if the file gets created (Settings saved). If not, you will have to debug it yourself, because I can't test without the hardware.

jdeltoft, 2017-11-24:
I tested out a fix on mac, and validated at least that doing a "pyinstaller main.spec" still worked on windows.  let me know what you think.

BetaRavener, 2017-11-24:
Which fix, the one I did or the one you are trying to merge. I first need you to try the one I created, because it doesn't require changing working directory.

jdeltoft, 2017-11-25:
Per my comments on that pull request, I deleted.  Thanks for the comments.  I will push the app main.spec changes once I figure out icons on mac.

BetaRavener, 2017-11-25:
Ok, thanks. And which icons do you mean? The one you can see in folder for uPyLoader, or those that are used in uPyLoader directly (e.g. 2 arrows on refresh ports button)?

jdeltoft, 2017-11-25:
I'm talking about the icon shown in finder in mac.  
[image]

BetaRavener, 2017-11-25:
Ok, I guess you will have to solve this one on your own :)

jdeltoft, 2017-11-25:
it's strange, because inside this app bundle, the Contents/Resources folder does indeed have main.ico that is your upy logo.  Maybe it's invalid for mac?

jdeltoft, 2017-11-25:
Works now on mac-dev branch.  closing issue

closed 2017-11-25 by jdeltoft