BetaRavener/uPyLoader #45: Add maximize and minimize buttons to terminal window

linusg, 2018-01-12:
Hi @BetaRavener,
this should be an easy one. First of all, thanks for this tool, works awesome and has more of an IDE-ish feel than working from the terminal.

One minor issue though:
[image]
As you see, there's only a close button, I can resize the window to my monitor's size but I'm used to just double-clicking the title bar.
Should be a matter of window flags only, even though my last PyQt project was a while ago..

BetaRavener, 2018-01-12:
Hi, sure this was easy to do and is implemented by a9c60f1. I hope you can run from source or build your own binary, because I don't think this warrants a new release. You can find necessary instructions in [Wiki](https://github.com/BetaRavener/uPyLoader/wiki). And I'm glad that you like the tool, it's always nice to have a feedback. :)

Closing this for now, if there's any issue, feel free to reopen.

closed 2018-01-12 by BetaRavener

linusg, 2018-01-12:
> I hope you can run from source or build your own binary, because I don't think this warrants a new release. You can find necessary instructions in Wiki.

Sure, I have the whole thing as a git repo, so a `git pull` was sufficient. Works like a charm, not that I would have expected different...

> it's always nice to have a feedback. :)

Being open-source "fanatic" and maintainer of a few repos myself I know that and therefore try whenever it feels appropriate. :slightly_smiling_face: 

This repo got a new watcher BTW, let's if I'll find for a contribution other than requests in the near future!