BetaRavener/uPyLoader #46: wifi connection not possible running asyncio script

balku, 2018-02-28:
HI,
I'm able to connect to webrepl running on ESP8266 (nodemcu-board) if connection is done via USB while my script using asyncio by loop.run_forever().
But, if I want to do the same using wifi, I get in 90% of my attemts an error message in the upper left corner of uPyLoader. Every task of my script does await asyncio.sleep(0).
Can I do something in my script to give webrepl a chance to interact with uPyLoader?
Many thanks
balku

BetaRavener, 2018-02-28:
You're stating that you can connect to WebREPL if you're using USB. That doesn't make much sense. WebREPL is only for WiFi connection. So if you are using COM port (Windows) or devtty (Linux), you're not using WebREPL. 

I don't know how much help I can offer. I never tried async on MicroPython. You can try using the [official WebREPL client] (https://github.com/micropython/webrepl). If that works for you without an issue, there's probably something in the communication that uPyLoader is not expecting.

balku, 2018-02-28:
Found a solution by import webrepl into my script and do a webrepl.start() after wifi connection is established. Now, all works fine. Both webrepl client (not using https) and uPyLoader.
Many thanks for your help.
balku

closed 2018-02-28 by balku

BetaRavener, 2018-02-28:
Great to hear that. Did you use daily build or something other than stable MicroPython binary release? I have found [this issue](https://github.com/micropython/micropython/issues/2432), which mentions that for everything else than stable release, WebREPL is turned off by default. 

This is possibly reason why you have to start WebREPL yourself. In such case I would recommend placing those 2 commands (`import` and `start`) at the end of boot script (`boot.py`), if you didn't do that already :). I will try to mention this issue in readme so other users won't get stuck.

balku, 2018-02-28:
I use micropython 1.9.3 stable release. Webrepl was not activated by default. I enabled it and provided a password. Without doing this, no connection is possible anyway.
Despite the activation in boot.py, I need to import and start it inside of my script. Only with this it works fine.