BetaRavener/uPyLoader #47: Upyloader stop working on serial connection!

cyberlab2018, 2018-04-03:
Hello Betaranever, sorry for bothering you again, today try to make the connection via serial port, upyloader stops working.
the connection via wifi works perfectly.
I want to tell you that if I use the previous version to the changes you made in report # 44 if the serial connection works well, apparently those changes affected the serial connection code.
Thanks greetings!
[image]

BetaRavener, 2018-04-03:
I'm too busy right now, so please use the previous version if you need to have serial working. I've at least looked at the code I was changind in #44, but I can't see anything that would affect Serial. I made modification to class that Serial uses, but these would only show up during file transfer, not when establishing connection. Hopefully I will be able to reproduce and fix it easily.

BetaRavener, 2018-08-09:
Closing this as an old one, but it may be that uPyLoader hung when connecting to board where program was running and capturing `KeyboardInterupt`. This was hopefully solved as part of #48.

closed 2018-08-09 by BetaRavener