BetaRavener/uPyLoader #49: Reset button

vpatron0, 2018-04-22:
Hi, can you add a reset button that sends Ctrl-C, "import machine; machine.reset()", and then uPyLoader disconnects (without causing error)?

It's just a convenience thing. I'd imagine a lot of people have their ESP code automatically run on boot and would like to reboot conveniently.

vpatron0, 2018-04-25:
I use a very simple workaround for now. I created a file called "machine_reset.py" with the contents below. I click on the file and click on the Execute button to reset.

```python
import machine
machine.reset()
```

BetaRavener, 2018-08-08:
Looking at implementing this now. Would just like to ask what's the reason for "and then uPyLoader disconnects"? I'd expect if you are resetting device, you want to continue working with it after it reboots.

BetaRavener, 2018-08-09:
Ah interesting, I have been fighting with the memory problem too. So `machine.reset()` works for that? Cool. Another thing you might try to prevent this is to put everything in function (call it main) so you don't create global variables. When the variables from function get out of scope, they should be collected. However as I said, out of memory error still happens to me sometimes. Might be libraries or really anything.

I'll look if the files can be flushed somehow and if I was going to do this, I would place it in `Menu->Action->Reset` so it's not the first thing you'll click as "newbie". Also a warning message might be good idea (with that checkbox "Don't show this again").

Or if we can find other way to get memory cleared up, it would be very nice.