BetaRavener/uPyLoader #50: uPyLoader doesn't execute

icr3at3, 2018-04-24:
# undefined symbol: drmGetDevices2
Whenever I try to execute the file after `chmod +x uPyLoader-linux` using `./uPyLoader-linux`, I get the following output - 
[image]

Please help me out here as I am trying to develop a Curriculum of 21st century skills and I am stuck in my cuz an Editor/IDE is a must for beginners.

vpatron0, 2018-04-25:
I remember I had similar issues. Several easy ways to make this work right. But the real problem is that the main.py is missing a proper python "shebang" line for Linux. Any of these will work:

Solution 1:
* Always type *python3 main.py* to run the program

Solution 2:
* Add the proper Linux shebang line as the very first line in *main.py*

```python
#!/usr/bin/env python3
```
Solution 3:
* Use a bash script to execute main.py. Create a file called uPyLoader with contents below. 

```bash
#!/usr/bin/env bash
python3 main.py
```
* Do *chmod +x uPyLoader* to make that file executable.
* You can now do ./uPyLoader to run the GUI

Or if you're using a Python virtual environment like you should, the bash script would be something like:
```bash
#!/usr/bin/env bash
source env/bin/activate
python3 main.py
deactivate
```

BetaRavener, 2018-04-25:
What you are describing is running from source. There might be some problems too (I'm not on Linux most of the time so not much testing on my side 😃  ), but the reported problem is tied to executable that was created with PyInstaller. In case you didn't find out yet, you can get the executable from [here](https://github.com/BetaRavener/uPyLoader/releases).

vpatron0, 2018-04-25:
Oh, I see. I didn't realize PyInstaller can be used to package Linux. Virtualenv might be more reliable for Linux instead of PyInstaller since tons of people use that. Still "running from source" but would package up PyQt, and all modules needed not part of baseline Python3. Linux comes with Python3 anyway, unlike Windows.

linusg, 2018-04-25:
> Oh, I see. I didn't realize PyInstaller can be used to package Linux.

It can basically create executables for every common and even not so common OS: https://www.pyinstaller.org/

> Virtualenv might be more reliable for Linux instead of PyInstaller since tons of people use that.
That's a different thing, you can't "ship" a virtualenv, users still have to manually create it and install the given list of dependencies into it. Basically one can do this already now, it's no difference at all.

vpatron0, 2018-04-25:
Ah, got it. Thanks linusg.

BetaRavener, 2018-08-08:
Closing this as a Pyinstaller issue. Fix doesn't seem to be in near sight, so users experiencing the issue are recommended [to run from source ](https://github.com/BetaRavener/uPyLoader/wiki/Running-uPyLoader-from-source) or [build their own executable](https://github.com/BetaRavener/uPyLoader/wiki/Development-notes).

closed 2018-08-08 by BetaRavener