BetaRavener/uPyLoader #54: Remember column witdth and sort sequence of Local files

robhamerling, 2018-07-09:
Version 0.1.3 displays local files on name in 'reversed' name-sequence and the name column is too narrow (for me). This can be changed by the user, but the adaptations are not remembered, so with every next startup this has to be redone.
It might be relevant to mention:
running uPyLoader under Linux Mint 19 from source: Python3 ~/uPyLoader-master/main.py

BetaRavener, 2018-07-09:
Please check that config file is saved next to the executable in folder and it is updated correctly (should change timestamp when closing uPyLoader). This feature was already implemented in #31, so it would be strange if it stopped working.

robhamerling, 2018-07-09:
The config.json file is updated: timestamp is changed and I can also see my changes in Options/Settings.
Linux Mint 19 is quite new and I experience this issue only recently, it may be since using Mint 19. 
I can hardly believe it, but under Mint 18.3 the issue doesn't manifest itself! No idea where to look for the difference, since otherwise uPyLoader works fine.

BetaRavener, 2018-07-09:
Could be really anything but I would blame it on Mint changing something that broke Qt support for this. Does other things like window sizes and positions get saved and restored?

robhamerling, 2018-07-10:
All other things but Name column width and file sort sequence look unchanged to me. 
I  tried to analyze the differences between the config.json files built with Mint18 and 19 (with kdiff3 after a json.load -> json.dumps operation).  I see some very minor value differences in main. But in localPanel there are significant value differences and it contains also several more values with Mint19 compared to Mint18. Would it be of help to attach these files to this issue? If yes, the original json files or my json.dumps() output?

BetaRavener, 2018-07-10:
You may try.. I don't know the internals of Qt nor Mint's window manager and I don't think it will be in my power to fix this, but we may at least find a culprit and then reference this in a ticket.

robhamerling, 2018-07-10:
I followed this scenario to create the attached formatted config.json files:

1. Mint 19: start uPyLoader without config.json
    Edit path and arguments of external router 
    Terminate and save config.json -> 19a
2. Mint 18.3: start without config.json
    Edit path and arguments of external router 
    Note: default File sort is descending Name sequence,  Name colum too narrow
    Terminate and save config.json -> 18a
3. Mint 18.3: start with config.json of step 2
    Increase window size, increase File Name column, change sort sequence to ascending
    Terminate and save config.json -> 18b
4. Mint 19: start with config.json of step 3
    Note: Windows size, File Name column width, sort sequences are same as with step 3!
    Terminate and save config.json -> 19b
5. Mint 19: start with config.json of step 4
    Note: Window size as in steps 3 and 4, File Name column width and sort sequence back to step 1.

The 4 saved config json files are formatted with the following Python3 script:

import json
l = ["18a", "19a", "18b", "19b"]
for f in l:
    fn = "/media/nas/uPyLoader-master/config" + f + ".json"
    with open(fn, "r") as fp:
        cfg = json.load(fp)
        fo = f + ".txt"
        with open(fo, "w") as fp:
            json.dump(cfg, fp, indent=4, sort_keys=True)

Below the results. Let's hope this helps to find the cause of the trouble.
Rob.

[18a.txt](https://github.com/BetaRavener/uPyLoader/files/2181601/18a.txt)
[18b.txt](https://github.com/BetaRavener/uPyLoader/files/2181602/18b.txt)
[19a.txt](https://github.com/BetaRavener/uPyLoader/files/2181603/19a.txt)
[19b.txt](https://github.com/BetaRavener/uPyLoader/files/2181604/19b.txt)

BetaRavener, 2018-07-10:
Just an explanation, what we are interested in is `_geometries->localPanel` list. Then there's second geometry - for main window, and the rest is settings that I created.

So looking at the files I can see that under Mint 19 it adds 4 more values at the end of the list. They are initialized to `0, 0, 0, 100` and after changing size they all get values `255`. Then there are changes in several other values. Overall the number of changes in v18 (between a and b) is smaller than in v19. I think I'm able to pinpoint what we are after: Probably a flag is changed at line 19 to change sort direction and lines 85 seems like size increase of the panel (same change happens in v19). 

Now the question is why there are more values in v19 and we need to determine what are those other changes (more so, what effect do they have). I would probably start by manually editing the config file and first removing the last 4 values under Mint 19. It might crash, in which case revert them to previous values and see what that does. If these last 4 values have no effect, you can check the other values that change between 19a and 19b (other than line 19 and 85) and try either reverting them or setting them to something else to see if that has any effect.

It will be quite tedious and might not yield results so it's up to you if it's worth your time. What I can do is check how exactly does geometry get saved into config file and try to look up those values somewhere in Qt sources, but that will also be pain in the ass.

BetaRavener, 2018-07-10:
I have found the function that is responsible for saving geometry of that panel (to be precise, it's header) and it calls [this function](https://github.com/qt/qtbase/blob/22e9ff9c164d9912fcbe7987f4d39e467b0a600f/src/widgets/itemviews/qheaderview.cpp#L3986) to write the data. The problem is that it doesn't match contents of the files at all, so I can't assign values to those fields. For example there should be some `visualIndices` vector saved which I would expect to be streamed as another array. The question now is why the difference.

BetaRavener, 2018-07-10:
Okay so I'm right about line 19, it's last byte of `sortIndicatorOrder`. And by some testing on my machine I'm sure that the line 85 is really the size of name column - again the last byte of it (I halved the width and this translated to 2 times smaller value). This should correspond to the `sectionSize` of the first section, which is the name column. So these values seems to save correctly also for Mint 19. There must be something else going on.

robhamerling, 2018-07-11:
I'm afraid further analysis is pretty much beyond my knowledge/experience level. 
If I were you I wouldn't spend much more time on it, certainly when I am the only one experiencing this phenomenon. But of course that is completely up to you, you might like the challenge or are eager to learn! ;-)  A solution would be nice, but I can live with the current inconvenience.
If you want me to do some more tests I'm happy help as good as I can.

BetaRavener, 2018-07-11:
I like the challenge but got really more important things to do :) At least for uPyLoader that would be directory support on the MCU. However, I can at least open a ticket in Qt system and let them know there's something weird going on, because it seems like their issue. 

One more thing you can do before that.. Try running the uPyLoader from source code and use the most recent PyQt. Maybe they fixed the problem while ago, but the uPyLoader executable is already 7 months old so it packs an old version. Here are the [instructions](https://github.com/BetaRavener/uPyLoader/wiki/Running-uPyLoader-from-source), it should be fairly easy.

robhamerling, 2018-07-11:
OK, but since I installed Mint 19 recently from scratch I think I have already the latest PyQt5:

> pip install PyQt5
Requirement already satisfied: PyQt5 in /usr/local/lib/python3.6/dist-packages (5.10.1)
Requirement already satisfied: sip<4.20,>=4.19.4 in /usr/local/lib/python3.6/dist-packages (from PyQt5) (4.19.8)

If I can do anything else let me know.

BetaRavener, 2018-07-11:
Well, you said you were using executable.. The executable isn't using PyQt from your system but has all dynamic libraries packed inside (that's why it's 25Mb and not just few Kb..). So in order to use PyQt you have installed in your system, you have to **run from source**. Basically calling `python main.py`.

robhamerling, 2018-07-11:
No! See my first msg (opening this issue):  I run uPyLoader from source.
But wait a little with further work on this. I'm suspecting my install of Mint 19.  I had a Mint19beta installed in VirtualBox and that does not show the issue! 
I have no free partition on my disk, and don't want to overwrite the existing Mint 19, so I'll install the official Mint 19 release under VirtualBox, add only the minimum required software to run uPyLoader and see what happens.
Keep your fingers crossed.....

BetaRavener, 2018-07-11:
Ah sorry, I totally overlooked that last line and since I saw you mentioning a version (which is assigned to exes) I automatically assumed that you are using executable.. my bad. Alright, holding my breath for that.

robhamerling, 2018-07-11:
No problem (but don't hold your breath for too long!). 
Good news: With the new Mint 19 system under Virtual Box uPyLoader works as good as I was used to before Mint 19: Name column width and File sort sequence are remembered properly. 
I'm afraid I have to conclude:
 - This issue must be qualified as 'user error'
 - I'll have to re-install Mint 19 
I appreciated your attempts to solve the issue, but my apologies for wasting your time!

BetaRavener, 2018-07-11:
It took just few hours so far so not much time 'wasted' :) It's unfortunate on your side but hopefully everything will work OK once it's done. Closing this for now, if you want to write followup or anything feel free to do so.

closed 2018-07-11 by BetaRavener