BetaRavener/uPyLoader #55: All file transfer fails

robhamerling, 2018-07-17:
I have been using uPyLoader in this case for a Sonoff S20 (esp8266 with 1 MB) for some time, but I suddenly got the message:

> Transfer scripts for UART are either missing or have wrong version.
> Please use 'File->Init transfer files' to fix this issue.

The _upload and _download files were listed, like boot.py and my own previously loaded files.
Removing all my own files didn't help.
Since _upload.py and -download.py were still present the File -> Init transfer files was grayed out.
I removed these files and executed Init transfer files, but that gives popup: Error: File transfer failed
So I decided to start from scratch: esptool.py erase_flash and write_flash (micropython 1.9.4). 
Now I see as only file boot.py, and I can start the File -> Init transfer files, but this still gives the File transfer failed popup.
I'm stuck.... 
Any suggestions to overcome this? Do you need more detailed info?

robhamerling, 2018-07-17:
The problem arose spontaneously (between two file transfers), so I feared for a hardware problem.
As alternative to uPyLoader I tried AdaFruit's ampy: file transfer works fine and my application works fine too again. So having reached lifetime of flash memory seems unlikely (I haven't used this device very intensively yet anyway).  uPyLoader still reports error when trying to send init transfer files.

Replacing the esp8266 is a no-go (at least for me): the S20 is a 'ready-to-use' device, see e.g.:
> https://www.itead.cc/smart-socket.html

I didn't succeed connecting via wifi. Never did that before and it doesn't want to connect automatically. Have to find out how. Because of the success with ampy this is probably not very opportune.

Tried file transfer to another esp8266 (nodemcu): no problems. But this 8266 still has the _upload and _download files which I didn't dare to remove!

BetaRavener, 2018-07-17:
If ampy works correctly, it's probably software issue indeed. I'll try to find a while to implement detailed error message. As I remember, you run from source code so you'll be able to pull those changes after I'm finished and it may give us better picture of what's going on.

profra, 2018-07-17:
Some idea ... try DOUT or DIO... not QIO settings during flash... I guess Sonoff is working with DOUT...

robhamerling, 2018-07-17:
Yes, I run uPyLoader from source. And yes, this S20 requires -fm dout. I started with dio as I was used to with my other esp8266s, and tried some other modes, but only dout gives a Python prompt!
The S20 has no UART connector but the internal PCB has place for a 4-pins .1" header (Gnd, Vcc, Rx,Tx), which I added. Google for it and you'll find several informational sites.
My S20s are meant to run as network stations connecting to a 'real' esp8266 as access point. I decided for an independent network (separate from my home network).  I  removed the 3 screws to open the enclosure and now use a rubber band to close it as long as I'm not satisfied with the application (will I ever be?). The possibility of loading programs via Wifi sounds attractive however!

BetaRavener, 2018-07-17:
Well you can't 'route' through another ESP, so you'd have to either temporarily add it to home network or start access point on each device. However, looking at the way S20 is configured, it seems like it should have access point already running (you should be able to connect to it through app, right?). It may be that the AP is hidden and not broadcasting SSID but I can't imagine how else the app would connect.

Anyway back to the issue, I'm just adding the error messages, so expect new commit in an hour or so..

robhamerling, 2018-07-18:
Downloaded the latest version of uPyLoader this morning (18 July).
Started with freshly flashed S20 (erase-flash then write_flash -fm dout Micropyhton 1.9.4). 
Strange: erase_flash took 0.0 seconds, in the past a few seconds. But write_flash seemed to work well.
At the start of uPyLoader and hitting 'Connect' I get the expected  'Transfer scripts problem' popup. When trying to load the transfer scripts I get an Error popup with
 >   Reason: Couln't locate transfer scripts.

Also strange: the Remote (MCU) filelist is empty, there should have been a 'boot.py'.
In a session with a terminal program I get the Python prompt, but when trying to get a filelist I see:
> >>> uos.listdir()
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
> MemoryError: memory allocation failed, allocating 2048 byt
> >>> 

And now also ampy doesn't work properly anymore!

So it seems you were right with suspecting a hardware problem with this S20.

About uPyLoader, S20 and Wifi: I'll try it out on another S20.

robhamerling, 2018-07-18:
[Somewhat offtopic] 
I fail to get a connection to an esp6266 via Wifi. My Laptop is connected to the AP of the esp8266 via Wifi (SSID = MicroPython-xxxxxx, laptop got IP 192.168.4.2) and in uPyLoader 'wifi' is selected (IP 192.168.4.1, port 8266), but when hitting 'Connect' I get a connection error.
I don't see what I'm missing or doing wrong.

BetaRavener, 2018-07-18:
Okay so that thing with transfer scripts is strange. I was also working on #51 yesterday so please check in settings that you have `Use transfer files` checked and `Use custom transfer files` unchecked. That should make uPyLoader use transfer scripts in the MCU folder. Btw, make sure that your working directory is the same as where `main.py` is located (e.g. don't run from your home directory using something like `python dir1\dir2\uPyLoader\main.py` but go to `dir1\dir2\uPyLoader` and run `python main.py`).

I will probably change the mechanism how path's are determined so that `main.py` location serves as an anchor but will have to check if that will also work with executables.

If it still fails, you can check `Use custom transfer files` and set `Path to custom transfer files` to `uPyLoader\mcu` folder where you should see the scripts (`upload.py` and `download.py`).

Somewhat offtopic: Everything seems correct to me. It should ask you for the password, not throw an error. Ah, now it occurred to me.. Since micropython ~1.9, the WebREPL isn't running by default. I think that's the problem. See this [guide](https://learn.adafruit.com/micropython-basics-esp8266-webrepl/access-webrepl) for setting up WebREPL. You will need UART for that though..

robhamerling, 2018-07-18:
Thanks for the explanations! 
The sad thing about the location of the transfer files is that I knew it and had it correctlly set previously. But my settings got lost after I switched to a new version of my OS (on another partition). 
I now made a note of it.  
Another sad thing is that transfer of these files failed to my probably defect S20. It gives a Python prompt in a terminal session but file operations fail.

After running webrepl_setup I could successfully communicate with the (good) S20 via Wifi! Looks very useful with 'closed' S20 devices.
BTW: Also running webrepl_setup fails on the defect S20!

BetaRavener, 2018-07-18:
Sorry to hear that you have so many issues, it really can make one frustrated over time.. Hopefully you can salvage the dead S20 in some way. Maybe replacing flash memory in this case would be worthwhile. 

On the other note it seems that your problem with transfer files is now fixed? It wasn't for nothing, those detailed error messages really make difference in debugging, I just needed a push :) If everything is fine now, I'll close the issue. I also created FAQ where I rewrote those instructions for future.

BetaRavener, 2018-07-18:
Thanks! FAQ updated and closing this for now. :)

closed 2018-07-18 by BetaRavener