BetaRavener/uPyLoader #56: Wifi connected but no files listed

robhamerling, 2018-07-19:
I'm trying to transfer files via Wifi.
I have run webrepl_setup on an esp8266 (actually a Sonoff S20), specified Enabled at boot and given a password.
From my laptop I made a wifi connection to this esp8266:  ifconfig shows as my IP: 192.168.4.3.
uPyLoader shows 'Wifi', IP 192.168.4.1 and port 8266 in the header. When hitting 'Connect' I am asked for the WebREPL password and after that uPyLoader shows as status: Connected.
The LED of the S20 shows me that the application which was running before the connection with uPyLoader is terminated (supposedly by Ctrl-C from uPyLoader), but otherwise nothing happens: no list of files in the esp8266, and no Error messages. Hitting 'List FIles' on the right half of the window blinks shortly but does nothing, also no error message. Nevertheless attempting to transfer a file gives a popup of the File Transfer widget with the progress indicator at 100%, but nothing arrived at the other end.
I read the info about Wifi and webREPL, but found no clue for this problem. What am I missing?

BetaRavener, 2018-07-19:
Try running `import os; os.listdir()` in terminal. What's the output?

robhamerling, 2018-07-19:
OK, uos.listdir() shows all the remote files.

BetaRavener, 2018-07-19:
Then there's problem somewhere in how the output is wait for or processed. There's still and open issue #48 which is related - when a program is running and it takes a while to finish, files are not listed. However, clicking the List Files button should list them. I'm packing for 2 weeks vacation so really don't have time to fix this now, but you may try to debug `list_files` procedure in `wifi_connection.py` if you know how to.. The procedure just sends the command I wrote above and waits for output which is parsed with simple regex.

robhamerling, 2018-07-19:
Thanks, I'll see what I can do. In the meantime: happy holidays!

robhamerling, 2018-07-28:
Tried to find the cause of the missing filelist, but to no avail.
I open uPyLoader.py from source in a terminal program (gtkterm), make connection and open its terminal window. When I hit the 'List Files" I see in the terminal window of uPyLoader the expected list of files between square brackets followed by the Python prompt. But in the gtkterm window I see only the Python prompt.
After I added a print statement in connection.py in the function read_to_next_prompt()  to show all the incoming bytes I see only arriving '\r\n>>> '  (without the quotes). So at that point the [list of files] has already disappeared, but I could not find why/where.

BetaRavener, 2018-08-06:
Hi, so I'm back and will be able to look into this maybe tomorrow. There's a function responsible for clearing terminal (without looking at code I think it's `read_junk`) that's probably eating the list of files.. I will have to think a bit how to make it more reliable, because with added timeouts it gets difficult to handle properly. Probably printing some special sequence just before the list would do the trick.

robhamerling, 2018-08-07:
Let me know if I can do something more to help.

BetaRavener, 2018-08-08:
Hi, I just released new code that should solve your issue. Please let me know if it works.

robhamerling, 2018-08-09:
I can confirm: the new version shows the list of remote files via the Wifi connection and seems to work otherwise as before (good!) too!

One remark: The use of a wifi connection looked ideal to me for the Sonoff S20s, because I wouldn't have to open the boxes for reprogramming. But it requires an active Access Point in each of the S20s. However I use the S20s as wifi stations and don't want all these APs to appear in a Wifi scan by whoever is in the neighbourhood. Therefore my application disables the AP.  And the app is otherwise rather simple, so I do not expect frequent updates.

robhamerling, 2018-08-09:
Thanks for the suggestion! A period with active AP at startup seems a good idea!

BetaRavener, 2018-08-09:
Not really.. I guess if you know how to fix it, you could just call `start()` in the script after activating AP? Or does it work only when you connect through UART?

robhamerling, 2018-08-09:
It appears there was something wrong with boot.py (I ignored the error message and everything seemed to work well regardless the message). After erase and flashing of the latest micropython no more error messages from boot.py and I was able to transfer files via Wifi (the termination of the application program with connect by uPyLoader worked too).  
Question: the terminal window of uPyLoader is not functional in Wifi-mode? Got no responses.

BetaRavener, 2018-08-09:
It is. Does anything appear when you send simple print command? You can try to send kill with Ctrl-C (the `-C` button in terminal). It should spit out the prompt (`>>>`). 

Otherwise it would require some debugging. It's possible that autoreader (thing that periodically reads device output and prints it to terminal) was not re-enabled after listing files. Would have to look into this if your problem won't go away.

robhamerling, 2018-08-09:
Following your suggestion I activate an AP in my application at startup/reset and deactivate it after a few minutes. Each of the S20s uses a separate SSID (via machine.unique_id()).  I make the AP active too when my application terminates due to an exception.  This looks like a workable and acceptable situation.

The terminal window of uPyLoader works indeed, although it appears and reacts slightly different than I was used to via UART. Entering a python statement in the lower half shows the echo and its result in the upper half. Also very useful is that I can reset the S20, thus restart the application with control-D!

uPyLoader becomes more and more useful to me (and hopefully to other users!).

BetaRavener, 2018-08-09:
"Entering a python statement in the lower half shows the echo and its result in the upper half." - should work the same for UART. Or have you been actually writing in the upper part whole time? (ouch..) If so, I need to make it more apparent that one should preferably use the lower half. The upper sends keystrokes directly to device thus doesn't allow edits before sending command and has some issues. Only thing I use the upper half for autocompletion using Tab (useful when I'm not sure about module or function name).

Anyway, glad that it worked. After your final reply I'll close the issue.

robhamerling, 2018-08-09:
I'm typing in the lower window!
Differences of Terminal window between Wifi and UART for example: 
- with Wifi hitting -C gives as only output  ">>>",  with UART much more text
- with wifi -D resets the device but gives no output at all in the terminal window.

jpbihin, 2019-03-08:
Hi, 
First : thank you for your job!  Very usefull and simple!
I re-open this thread because I have the same problem but I canot solve it ...
"no list of files in the esp8266 when I connect via wifi" . My OS is Windows 10.
I read (8th august) : "I just released new code that should solve your issue. ". I guess you talk about the 0.1.4 release ? When I upload it,  I get the 0.1.3 and the issue is still there..
Can you help me solve this ?
Sorry for my poor english (frenchspeaking..)
Regards,
Jean-Paul