BetaRavener/uPyLoader #57: Allow domain names instead of IP address

vpatron, 2018-08-09:
Hello BetaRavener,

Feature request: can you allow any domain name instead of forcing us to use IP addresses?

For newer routers, and routers using firmware such as dd-wrt, openwrt, tomato, you can set a local domain name and clients can be accessed by domain names instead of by IP address, but uPyLoader only allows IP addresses. IP addresses are terrible because they can change (unless forced static).

For example, at home I run Tomato firmware on my router and set my domain to "bgw.lan", so if my ESP8266 is called "esp-patio", I can refer to it as "esp-patio.bgw.lan" and the router takes care of routing it. In fact, I can just refer to it by "esp-porch" without the local domain and it knows to route it locally.

So would be cool if uPyLoader allows not just "x.x.x" or "x.x" domain name or IPs, but really allow any name "x" and just trap the error. Thanks!

BetaRavener, 2018-08-09:
Ok, no problem, it's easy to resolve domain names to IPs in python (maybe it will even work just with the domain name). I didn't know about such feature on routers and most of time time I connect to AP, so IP is always the same. But since you have use for this, why not..

vpatron, 2018-08-14:
Hi BR, it's working great!! Here's what I tested:

1. I can connect to `esp-lv_room` or `esp-lv_room.bgw.lan` (just the machine name or full domain name) or the IP address.
1. It now stops a running script that is programmed to autorun on boot
1. It resets/restarts the ESP when I disconnect the wifi connection
1. It no longer resets the attached NodeMCU board even though I am connecting to a different board over wifi.

This is now working very smoothly! Thank you so much!

By the way, if you're interested, Tomato router firmware works very nicely and is easy to use, [regular tomato](http://tomato.groov.pl/?page_id=69) for basic and older routers or [advanced tomato](https://advancedtomato.com/) for fancy routers.

-Vince

vpatron, 2018-08-14:
Oh, one thing that still doesn't work like before: when I connect via serial to a NodeMCU, it resets it and it cannot stop the autorunning program. Even if I use a different terminal program to first stop the ESP, connecting via serial resets/restarts the autorun program. But this is a different issue. For WiFi connection, it works perfectly!

BetaRavener, 2018-08-14:
Thanks for trying out and recommendation about the firmware. This was only supposed to allow hostname rather than just IP in Address so this issue can be closed. The reset thing will be resolved in #48, although it's getting tricky to figure out.

closed 2018-08-14 by BetaRavener