BetaRavener/uPyLoader #58: Allow non-Python files to be edited

vpatron, 2018-08-14:
Hello BR, the built-in editor only allows .py files. This seems overly restrictive. There are lots of useful text files other than Python files.

I like using JSON files for configuration, such as hostname of ESP board, name of AP to connect to, and AP password instead of hard-coding into the Python program.

Can you let the editor edit these other files? 

To prevent editing binary files, perhaps you can read say the first 1k of data to check for only printable characters to determine if it's a text file. See [StackExchange](https://stackoverflow.com/questions/3636928/test-if-a-python-string-is-printable)

BetaRavener, 2018-08-14:
You're right that it is overly restricting and the idea with printable characters would probably work nicely. It should be quite easy to implement so I'll probably address this later today.

BetaRavener, 2018-08-28:
Sorry that I didn't get earlier to this, should be working now.

closed 2018-09-18 by BetaRavener