BetaRavener/uPyLoader #60: Problems in the Windows environment

profra, 2018-09-07:
I have not worked with uPyLoader for a long time and I have started with a new installation (git clone) and wanted to work from source (python3 main.py). I have encountered an error I can not resolve, I am not a tried pythonist... see the picture
[image]
This error only appears in the WIN environment, in the NIX environment is everything OK.
Have you any idea what to do?

profra, 2018-09-07:
Please, can you look at this listing if you see something suspicious?
[image]

BetaRavener, 2018-09-07:
Nothing really. You have both `pyserial` and `PyQt5` although in higher version than me. I was suggesting the reinstall because maybe something got corrupted.

profra, 2018-09-07:
FYI ... after several attempts to resolve this problem, I had to uninstall PyQt5 and PyQt5-sip. I tried installing version 5.6.0 (according to yours), but pip3 found version 5.7.1. as the oldest.

[image]

After these steps, uPyLoader started working.  I remember the last thing I installed in Python, the mu-editor, which is a very simple editor for MicroPython and CircuitPython, greatly supported by Adafruit ... that probably did upgrade of PyQt5 to the latest version, which is probably not compatible with uPyLoader. Maybe this is good information for you, because even other people may notice a similar problem.

BetaRavener, 2018-09-07:
I got PyQt 5.6 from Anaconda package where it seems to be most recent. Thanks for info, if someone else comes up with the problem I'll download new PyQt and see what's the issue. For now I suppose it was individual.

closed 2018-09-07 by BetaRavener