BetaRavener/uPyLoader #61: private config file desired

robhamerling, 2018-11-06:
I have most computer material on a NAS, including uPyLoader  In my system setup I use an identical directory structure (same base names) in all (Linux) system so I can use the same path specification for files on the NAS with all systems. Consequently most settings for uPyLoader are the same and a common/shared config file is OK.
However on the Raspberry I want to use a different source editor than on my desktop PC and I need for the Raspberry a different cross compiler binary as well.  In this situation a single uPyLoader config file is not ideal! A solution for this situation would be to have private config files (probably in the home directories). 
Rob.

BetaRavener, 2018-11-08:
Hi Rob,

I just want to let you know that I've seen this issue and will try to solve it. However, do you require both that:
* In most cases common configuration is used (e.g. that from NAS)
* In some cases you can use separate config file

or you're okay configuring uPyLoader for each machine. Because I can see some problems if both of those behaviors need to be supported.

robhamerling, 2018-11-08:
I've thought about a combination of a common and  private config files. It may not be too difficult for uPyLoader to read one after the other, but it requires a facility for the user to indicate if a change applies to the common or to the private config file. And uPyLoader must update one or both configs depending on the user-input.

Me personally would be happy with only private config files (and consequently having a dedicated config file on each system from which I invoke uPyLoader). When I view all the hidden directories and files in my home directory I think uPyLoader may even be the exception by having its config file in the directory of the binary executable!  

You might consider giving the user a choice between common of private at first startup of uPyLoader when it cannot find a config file (assuming there is no default config file initially).