BetaRavener/uPyLoader #62: External editor

DanielIbaseta, 2018-11-30:
The external editor doesn't seem to work in Ubuntu, ¿How is the reference to the loaded file passed to the "External editor arguments"? "%f" doesn't seem to work. Also, I'm passing the correct path for the editor, and it just don't show up empty or give an error, it just opens the built-in code editor.

BetaRavener, 2019-01-27:
Sorry didn't get notified on this issue. The external editor is started through Python API `Popen` which handles all the argument passing. You can see how the wildcard is replaced with file path here: https://github.com/BetaRavener/uPyLoader/blob/master/src/gui/main_window.py#L574. 

The internal editor is opened only in case the "Path to external editor" is left out empty. As long as there is something (not even correct path) entered, it should attempt to launch anything with that path. You could try to debug and see why it goes into wrong branch of code.

BetaRavener, 2019-06-03:
Closing due to inactivity.

closed 2019-06-03 by BetaRavener