BetaRavener/uPyLoader #64: reopening issue #56 no list of files with wifi connection

jpbihin, 2019-03-08:


jpbihin, 2019-03-08:
Hi,
First : thank you for your job! Very usefull and simple!
I re-open this thread because I have the same problem but I canot solve it ...
"no list of files in the esp8266 when I connect via wifi" . My OS is Windows 10.
I read (8th august) : "I just released new code that should solve your issue. ". I guess you talk about the 0.1.4 release ? When I upload it, I get the 0.1.3 and the issue is still there..
Can you help me solve this ?
Sorry for my poor english (frenchspeaking..)
Regards,
Jean-Paul

BetaRavener, 2019-03-08:
Hi @jpbigin. First of all, please make sure that the connection to ESP actually works by opening a terminal (view->terminal) and trying some random command like `print("Hello")`.. if it does work, try to manually do `import os; os.listdir()` and copy output here. You can also try hitting the "List Files" button, sometimes it just misbehaves on first try.

If you can't establish communication between ESP and terminal, there might be multiple problems, for example running script that can't be killed. In that case I would probably suggest flashing ESP with firmware again just to be sure there isn't anything wrong going on in it.

jpbihin, 2019-03-08:
Thanks !
In wifi mode, the esp responds correctly. Also os.listdir(). 
When I hit "list files", the list of files appears in the terminal but not in the main window.  
[image]
After hiting "list files" several times, the list finaly appears but randomly, after at least 5 or 6 hits.
I run the 0.1.3 version of Upyloader (menu <help><about upyloader>). When I download the 0.1.4 , I get a 0.1.3 version.
Regards, 
Jean-Paul

BetaRavener, 2019-03-08:
Hi again..
yes I have just tried downloading version 0.1.4 and it shows 0.1.3 but I just forgot to change that when releasing new executable. It should be the newest version. Anyway, the executables are somewhat old. Could you please try to run uPyLoader from source code? Instructions here: https://github.com/BetaRavener/uPyLoader/wiki/Running-uPyLoader-from-source

If the problem persists, I will have to investigate more.. you are probably receiving some unexpected format of the string (although from image it looks okay), maybe something is missing at the time it is being processed, maybe the message wasn't fully received. Could be many things.

jpbihin, 2019-03-09:
Hello, 
It works fine with the source code ! :o)
And thanks again for Upyloader. I do really like it.
Jean-Paul

BetaRavener, 2019-03-09:
Thanks for letting me know. I will try to rebuild executables and reupload them, just in case I really attached the old ones. :) And glad that you like the tool, it's always nice to hear feedback.

Will keep this open for now and I will notify you when I uploaded executables so you can try it out again.