BetaRavener/uPyLoader #65: Currently unusable on Mac due to the colors/theme, pref port action also a bit off

jdeltoft, 2019-04-22:


jdeltoft, 2019-04-22:
Also, when I start uPyloader when my preferred port isn't yet available, some other port is selected.  But then this code was preventing my preferred port from having any action when I then turned on my nodemcu which was really the whole point of preferred port.

[image]

I updated it to use preferred port any time it's NEWLY found.

BetaRavener, 2019-04-22:
I'll need to see about approving that dark theme PR - if it changes color scheme globally for everyone, not just those using dark theme, it would cause the same problems for other people.

I wouldn't change preferred port behavior either. I can leave it open for discussion but that setting is meant mostly as start-up setting, not port auto-detection. It's disturbing to workflow if port always changes to the preferred one if you are working on a device that is not on that port (say you normally use wifi but want to debug using UART for a while). 

There's nothing wrong modifying uPyloader to your needs, I'm just not sure if that aligns with needs of majority of users.