BetaRavener/uPyLoader #67: Mac 100% crash on closing of terminal (editor)

jdeltoft, 2019-04-22:
Whenever I close the terminal window (I have to use the 'x' in the upper left, since I don't see any other "close" button) I get 100% segmentation fault 11 in python.  Here is the trace:

Process:               Python [4120]
Path:                  /usr/local/Cellar/python/3.7.1/Frameworks/Python.framework/Versions/3.7/Resources/Python.app/Contents/MacOS/Python
Identifier:            org.python.python
Version:               3.7.1 (3.7.1)
Code Type:             X86-64 (Native)
Parent Process:        bash [1384]
Responsible:           Python [4120]
User ID:               546744075

Date/Time:             2019-04-22 07:22:01.804 -0500
OS Version:            Mac OS X 10.14.4 (18E226)
Report Version:        12
Anonymous UUID:        F7317033-88BF-51E1-4146-50BA5FD0ACD1

Sleep/Wake UUID:       D02D129A-4D4D-42B8-B874-C92C7709FFA1

Time Awake Since Boot: 200000 seconds
Time Since Wake:       84000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [4120]

Application Specific Information:
Calling windowShouldClose: on the delegate for the QNSPanel 0x7f9addf7a090

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   QtGui                         	0x00000001083ed36a QWindow::event(QEvent*) + 538
1   QtWidgets                     	0x00000001097baf8f 0x109752000 + 429967
2   QtWidgets                     	0x0000000109761d32 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 306
3   QtWidgets                     	0x000000010976310d QApplication::notify(QObject*, QEvent*) + 573
4   QtWidgets.so                  	0x00000001093317ca sipQApplication::notify(QObject*, QEvent*) + 234
5   QtCore                        	0x0000000108aa085f QCoreApplication::notifyInternal2(QObject*, QEvent*) + 159
6   QtGui                         	0x00000001083dd1e0 QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) + 2368
7   QtGui                         	0x00000001083c5d4b QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 155
8   QtGui                         	0x00000001083c17b0 QWindowSystemInterface::flushWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 576
9   libqcocoa.dylib               	0x000000010cffccdd QCocoaWindow::windowShouldClose() + 205
10  libqcocoa.dylib               	0x000000010d00d8eb -[QNSWindowDelegate windowShouldClose:] + 27
11  com.apple.AppKit              	0x00007fff2ca4d7ec __19-[NSWindow __close]_block_invoke + 184
12  com.apple.AppKit              	0x00007fff2ca4d727 -[NSWindow __close] + 292
13  com.apple.AppKit              	0x00007fff2c963e80 -[NSApplication(NSResponder) sendAction:to:from:] + 312
14  com.apple.AppKit              	0x00007fff2c9ce196 -[NSControl sendAction:to:] + 86
15  com.apple.AppKit              	0x00007fff2c9ce0c8 __26-[NSCell _sendActionFrom:]_block_invoke + 136
16  com.apple.AppKit              	0x00007fff2c9cdfca -[NSCell _sendActionFrom:] + 178
17  com.apple.AppKit              	0x00007fff2c9fad4f -[NSButtonCell _sendActionFrom:] + 96
18  com.apple.AppKit              	0x00007fff2c9cc8e5 -[NSCell trackMouse:inRect:ofView:untilMouseUp:] + 2375
19  com.apple.AppKit              	0x00007fff2c9faaa0 -[NSButtonCell trackMouse:inRect:ofView:untilMouseUp:] + 698
20  com.apple.AppKit              	0x00007fff2c9cb322 -[NSControl mouseDown:] + 791
21  com.apple.AppKit              	0x00007fff2ca3515d -[_NSThemeWidget mouseDown:] + 86
22  com.apple.AppKit              	0x00007fff2c8a716f -[NSWindow(NSEventRouting) _handleMouseDownEvent:isDelayedEvent:] + 5724
23  com.apple.AppKit              	0x00007fff2c7dd9de -[NSWindow(NSEventRouting) _reallySendEvent:isDelayedEvent:] + 2295
24  com.apple.AppKit              	0x00007fff2c7dce9f -[NSWindow(NSEventRouting) sendEvent:] + 478
25  libqcocoa.dylib               	0x000000010d00cc70 -[QNSWindow sendEvent:] + 528
26  com.apple.AppKit              	0x00007fff2c67c6c3 -[NSApplication(NSEvent) sendEvent:] + 331
27  libqcocoa.dylib               	0x000000010d01326a -[QNSApplication sendEvent:] + 74
28  com.apple.AppKit              	0x00007fff2c66aee8 -[NSApplication run] + 755
29  libqcocoa.dylib               	0x000000010d00f8ed QCocoaEventDispatcher::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 2397
30  QtCore                        	0x0000000108a9c3b2 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 418
31  QtCore                        	0x0000000108aa0f72 QCoreApplication::exec() + 402
32  QtWidgets.so                  	0x0000000109421662 meth_QApplication_exec_(_object*, _object*) + 82
33  org.python.python             	0x0000000107b4720f _PyMethodDef_RawFastCallKeywords + 236
34  org.python.python             	0x0000000107b468af _PyCFunction_FastCallKeywords + 44
35  org.python.python             	0x0000000107bdcb2b call_function + 636
36  org.python.python             	0x0000000107bd5771 _PyEval_EvalFrameDefault + 7016
37  org.python.python             	0x0000000107bdd432 _PyEval_EvalCodeWithName + 1835
38  org.python.python             	0x0000000107bd3b81 PyEval_EvalCode + 42
39  org.python.python             	0x0000000107c026ed run_mod + 54
40  org.python.python             	0x0000000107c01708 PyRun_FileExFlags + 164
41  org.python.python             	0x0000000107c00dc2 PyRun_SimpleFileExFlags + 266
42  org.python.python             	0x0000000107c18adc pymain_main + 5665
43  org.python.python             	0x0000000107c190ff _Py_UnixMain + 75
44  libdyld.dylib                 	0x00007fff5b4633d5 start + 1

BetaRavener, 2019-04-22:
When did this start happening? Did you do system update? Can't tell what's going wrong from the stack trace as I have 0 experience with OSX, There's some related information here: https://github.com/rust-windowing/glutin/issues/778 . Strange that it happens only for those 2 windows,. file editor and terminal right? Did you try settings dialog for example?

jdeltoft, 2019-04-23:
Settings did not do this.  I only saw it for editor.

BetaRavener, 2019-04-27:
Sorry, doesn't have lot of time lately.. by editor you mean A (https://raw.githubusercontent.com/BetaRavener/uPyLoader/master/images/editor.png) or B (https://raw.githubusercontent.com/BetaRavener/uPyLoader/master/images/term.png). Anyway it might be problem that OSX now doesn't like when there are connections when the window is closed. You could try to disconnect all Qt signals before closing window and connect them again on reopen. I unfortunately won't be able to test it for you as I have no Mac machine but if you need assistance just ping me here.

BetaRavener, 2019-06-03:
Closing this as unable to reproduce. If there is more information, please feel free to reopen or create new issue.

closed 2019-06-03 by BetaRavener