BetaRavener/uPyLoader #69: Test Editor cannot locate a file that's on the MCU

laurentopia, 2019-05-25:
[image]

BetaRavener, 2019-05-25:
It's actually telling you that the local file path is wrong. You were trying to edit the MCU `boot.py` file, correct? Can you tell me which actions you took - where you clicked etc.?

laurentopia, 2019-05-25:
So the text editor only edits text that's on the computer?

BetaRavener, 2019-05-25:
No, it should be also able to do it for MCU files, that's why I was asking what actions you took so I can reproduce the problem. I'll try to check it tomorrow..

laurentopia, 2019-05-26:
I selected the .py file from the MCU (a maix bit) the clicked on editor in the menu, then nothing happened so I entered the name of the py file in the field as you see and pressed enter. nothing I think, so I pressed save and that's when the error message appeared.

BetaRavener, 2019-05-26:
Try double-clicking the file in the list.. The editing happens on the PC so the file needs to be transferred first. Just writing name into the field won't do anything and pressing save would save empty file with that name to MCU.

laurentopia, 2019-05-27:
Yes! that works. Thanks.

closed 2019-05-27 by laurentopia