BetaRavener/uPyLoader #72: What is the required list of  built-in modules in Micropython to use uPyloader?

wuhanstudio, 2019-06-03:
Hello, recently I'm trying to use micropython on [rt-thread](https://github.com/rt-Thread/rt-thread/), a burgeoning RTOS in China, through serial connection. Micropython has been released as a [package](https://github.com/RT-Thread-packages/micropython) on rt-thread and it works for now, while it failed to "Init transfer files" when I tried to use uPyloader to upload files, though "List Files" did work.

I think some modules required for mpy to communicate with uPyloader my be missing but I cannot figure it/them out.

Here's the list of available packages:

- System Modules

    uos, _thraed, uselect, uctypes, uerrno, ffi

- Tools Modules

    cmath, ubinascii, uhashlib, uheapq, ujson, ure, uzlib, urandom

- Networkd Modules

    usocket, ussl

Could you please help me figure out what packages have to be enabled? Thanks sincerely for your time.

wuhanstudio, 2019-06-03:
With **uos** and **ubinascii** enabled, I can do: 

- List Files
- Execute
- Remove
- Transfer

But still download is not working

[image]

BetaRavener, 2019-06-03:
I've looked at the differences between `__upload.py` and `__download.py` since former is working while later is not, but there are not many differences. I suppose that `ubinascii` has symmetric functions so if `a2b_base64` is supported, `b2a_base64` is also. In that case, there's just line 25 & 26 which might cause issues: https://github.com/BetaRavener/uPyLoader/blob/master/mcu/download.py#L25 .

I can't try to make it work for you as I lack the device, but you can modify the transfer scripts by commenting out lines that seem problematic until you pinpoint the issue and the script executes (it may produce error later on due to changed behavior but that's fine). You can use `Path to custom transfer scripts` in the Settings to upload your edited transfer scripts or if you are running from source, simply editing scripts in `mcu` folder will work.

wuhanstudio, 2019-06-04:
Thanks a lot for your help. It's lucky that the working one is upload which makes it easier to upload/test scripts.

If I manage to find a solution later, I'll post it here. Thanks again for your suggestion.

closed 2019-06-04 by wuhanstudio