BetaRavener/uPyLoader #74: NUCLEO-F767ZI on Mac OS

sungkhum, 2019-07-29:
I'm having trouble getting file transfer to work on a NUCLEO-F767ZI on my Mac. Do you have any pointers? I can connect via REPL fine (screen /dev/tty.usbmodem1413 115200) and connect in uPyLoader (shows a list of files on the F767ZI), but always says the UART transfer files are missing.

I can connect in Windows 10 and transfer files. But on my Mac (even though the transfer files are there because I added them on my Windows machine) I get the UART error. I tried using a custom path for the transfer files, but that didn't help.

When I try to "Init transfer files" the program hangs.

Any thoughts?

BetaRavener, 2019-07-31:
I remember there was once an issue with privileges on the TTY device, you could try checking that but since the REPL works, this doesn't seem probable. I'm sorry but I don't have any other ideas and the best way would be to debug the file transfer routine (or at least the part where it checks the files).

BetaRavener, 2019-11-19:
Closed due to inactivity

closed 2019-11-19 by BetaRavener