BetaRavener/uPyLoader #75: upyloader can not run on win32(x86) system

yellowway, 2019-11-18:
as the title ! or which version is suitable for my old Win32 PC? THANKS!

BetaRavener, 2019-12-26:
Hi, sorry for late reply. Your best bet currently is to run directly from source. I don't have Win32 system anymore to create the package on. I wasn't aware that Win64 package is unable to run on those systems. I might look in the future on github CI options and if it would be possible to make this automatic, but it will probably be too late for you.

Ps: ofc, after you manage to make it run from source, you can create package for yourself, although at that point it might be useless.

BetaRavener, 2024-03-08:
Won't be able to test this. please use more recent OS. At very least, it should always be possible to run 64bit Linux OS in VirtualBox on top of Win32 system, though setting up USB passthrough for serial may be painful.

closed 2024-03-08 by BetaRavener