BetaRavener/uPyLoader #76: I cannot use vs code as External code editor.

talking-toaster, 2019-12-12:
I tried change `Optiond>Settings>Editer` .I used `E:/VS code/Microsoft VS Code/Code.exe` as first field .In the second filed ,I tried `%f` and ``but they do not work.

BetaRavener, 2019-12-26:
Hi, sorry for late reply. I'll try to look into this soon. Could you in the meantime try delete spaces in your path to vs code? It's a long shot (I think that I normally use Program Files folder), but might help.

BetaRavener, 2024-03-08:
Might be the fact that it's 4 years later, but can't reproduce, works as expected:
[image]

opens the file in VS Code, even if no Code instance was open.

closed 2024-03-08 by BetaRavener