BetaRavener/uPyLoader #77: Transfer files to PC

ChahidTALHA, 2020-08-13:
I have an issue about transferring files to the PC.
Everything works fine connecting my sipeed Maixduino board, files appear as wanted. the problem comes when i want to transfer files in the flash to my PC. a pop up dialogue called "File Transfer" pretending it is "reading file" and stuck in that stage. for over than 15mins.
Can i get some help?
[image]

BetaRavener, 2020-08-13:
Don't know why you changed the picture, but from the original post I see some strange filenames on your MCU that look like paths. Directory support was never fully implemented so if you somehow managed to transfer those files into folders, it's likely that reading them back won't work.

Try using simple names without special characters to start with. You could also check terminal for any errors.

ChahidTALHA, 2020-08-17:
Hi again, 
those strange filenames have nothing to do with the file transfer.
Instead, I fixed the problem after re-reading the documentation, I forget to "Init files transfer" once i connect my Maixduino board. 
Problem fixed and I thank you for your time @BetaRavener

BetaRavener, 2020-08-17:
Ok, glad you figured it out!

closed 2020-08-17 by BetaRavener