BetaRavener/uPyLoader #78: I cant connect with my Modul ESP8266MOD

aredasoufian, 2020-08-30:


BetaRavener, 2020-08-30:
Please provide description on how does your setup look like (wifi or uart? etc..), what you tried, where did it go wrong.

aredasoufian, 2020-08-30:
I installed the drivers and it came out in COM3 port by USB cable I tried to connect but nothing several attempts can you help me please

BetaRavener, 2020-08-30:
If it's not getting into "Connected" state, you may need to flash it: https://github.com/BetaRavener/uPyLoader#flashing-firmware-only-for-esp8266

BetaRavener, 2020-08-30:
Here's firmware page: https://micropython.org/download/esp8266/

aredasoufian, 2020-09-01:
thank you brother

BetaRavener, 2020-09-02:
Let me know if the issue is resolved so I can close it, or take next step..

aredasoufian, 2020-09-04:
yes brother is already resolved thank you vey much

aredasoufian, 2020-09-04:
Yes brother, case closed, thanks for your help

closed 2020-09-09 by BetaRavener