BetaRavener/uPyLoader #79: Hangs when folder transfer is attempted

Quadrangle, 2020-11-09:
In versions 0.13 and 0.14, when I select a folder in the Local window, click on Transfer, the File Transfer dialog appears with "Saving file."  The progress bar does not move.  The Cancel button can receive focus, but does nothing.  The entire program is frozen.  I've been using Task Manager to kill it.

But, keep up the good work!  I had a recent heavy session with some PyCharm people and told them they should wish to be as good as uPyLoader.

BetaRavener, 2020-11-09:
Hi. What connection type are you using, serial or WiFi? 

If serial: just to make sure, did you upload transfer scripts as noted in Readme? The most usual cause for this in past was either the scripts or filesystem getting corrupt for one reason or another. uPyLoader should check the version of scripts when connecting and complain if they are missing or there's version mismatch, but there's not a full file check like MD5 or CRC.

And thanks for flattering words :)

BetaRavener, 2021-02-04:
Closing due to inactivity.

closed 2021-02-04 by BetaRavener