BetaRavener/uPyLoader #86: Problem: No connection with ESP32S2

robhamerling, 2022-02-03:
uuPyLoader (0.1.3) fails to connect to boards with an ESP32-S2, in my case:
- LilyGO TTGO T8 ESP32-S2
- Wemos S2 mini ESP32-S2

No problems with loading firmware (with esptool.py).
The first board has dipswitches to choose between 'USB' and 'OTG' with which the board appears as ttyUSB0 or ttyACM0. uPyLoader detects these correctly but reports connection error with both.
Thonny and MPRemote work fine.

BetaRavener, 2022-02-03:
I don't have any ESP32-S2 board unfortunately, but there's not a lot that can go wrong for serial connection. Either it fails to open the port, or cannot read prompt (`>>>`). You can find the code that setups connection [here](https://github.com/BetaRavener/uPyLoader/blob/master/src/connection/serial_connection.py#L17). Can you check that every line starts with this sequence? Playing with `dtr` and `rts` values may also lead to some discovery, though for all boards so far the current `False` and `False` worked fine.

robhamerling, 2022-02-03:
While in a serial-terminal session (with GTKTerm) I already noticed a different behaviour (compared to a 'classic' ESP32) with pushing the reset button:  no output, no new prompt! 
Toggling DTR or/and RTS does not result in a reset and a new prompt, also different than classic ESP32. However a 'manual' Ctrl+D does result in a soft reboot with at the end of the serial output a >>> prompt.
After changing in 'serial_connection.py' both serial.dtr and serial.rts to **True** (lines 25,26) uPyLoader successfully connects again to both my ESP32-S2 boards. Then I tried this modified code with a classic ESP32: connects fine too!  
To be complete I tested with firmware:
MicroPython v1.18 on 2022-02-02; LOLIN_S2_MINI with ESP32-S2FN4R2
(in the serial termnal session it does not show, but I know it is 'unstable-v1.18-68')

BetaRavener, 2022-02-03:
Thanks for trying this out and glad something worked out. Can you explain bit more how are these two related?

> Toggling DTR or/and RTS does not result in a reset and a new prompt"

> After changing in 'serial_connection.py' both serial.dtr and serial.rts to True (lines 25,26) uPyLoader successfully connects again

Does solely doing the second fix your issue? How did you try toggling DTR and RTS in the first quote?
I might just add two checkboxes to the UI so people can easily change the values if some boards use less standard approach.

robhamerling, 2022-02-04:
A connect of uPyLoader to a ESP32S2 is only successful when both DTR and RTS are set True in serial_connection.py (no connect when one or both are set False).  This is the only change needed for a successful connect.

A 'classic' ESP32 seems insensible for the setting of DTR and RTS:  when both are set True or False it connects successfully. So when you decide to make these settings configurable I would think True would be the preferred default setting.

Toggling DTR and RTS is a feature of serial terminal emulator GTKTerm. When I toggle one or both there is no visible reaction on the screen with an ESP32S2. However a 'classic' ESP32 reacts to a transition of DTR from False to True with a POWERON_RESET (no reaction of DTR from True to False, no reaction at all when toggling RTS).

BetaRavener, 2022-02-04:
Thank you for details. I'll test with ESP8266 later and if it behaves well, I make RTS=True DTR=True new default