BetaRavener/uPyLoader #87: No more file transfer progress display after upgrading from Mint 20.3 to 21

robhamerling, 2022-08-20:
After upgrade from Linux Mint 20.3 to Mint 21 uPyLoader file transfer doesn't work anymore. At least it looks like that at first sight, but it seems that only the file-transfer progress bar remains 'empty'. Same behaviour on a newly installed Mint 21 on a different PC. 
I guess this might be a QT issue.  So no big deal, but annoying....

BetaRavener, 2022-08-22:
Hi, do you mean that it transfers the file, but the progress bar appears empty the whole time? Or that the operation never finishes, and the file is never sent over. 

Based on "no big deal", it sounds like the latter, but would like to understand this better.

robhamerling, 2022-08-22:
Sorry that I wasn't clear enough...
Files are transferred correctly, and everything else seems to work fine. 
Only the progress bar remains empty (shows 0% all the time during the file transfer), but closes normally after the file transfer completed.

BetaRavener, 2023-01-18:
Sorry for the (considerable) wait. I've installed Mate 21.1 with Cinnamon in VirtualBox and did quick hacks to trigger transfer dialog without MCU around. Reproduced and spotted the issue right away from console logs. Somewhere along Qt started expecting (or checking) an `int` as input to the progress bar. It's one line fix I'll merge tomorrow.

closed 2023-03-07 by BetaRavener